RoboticsBrno/ServoESP32 #1: Order of parameters in attach

s00500, 2018-04-16:
This library changes the order of parameters in the attach() function as compared to the standard arduino library, please change this or document it!
Thanks

JarekParal, 2018-04-16:
Yes, that is my fault. I'm probably creating a new interface for the function attach() with two options:
1) compatible with Arduino servo library - without parameter channel -> channel will be set automatically
2) parameter channel change from int to enum class and then the standard function overloading will work 

What do you think about this change?

s00500, 2018-04-17:
yes perfect, I was also thinking about having the channel as an optional attribute :-)

JarekParal, 2019-03-01:
Actually, I improve the documentation and in the future, I would like to refactor the library with cooperation to #2.