RoboticsBrno/ServoESP32 #9: Naming conflicts with "default" <Servo.h>

platisd, 2020-01-15:
When compiling for the `DOIT ESP32 DEVKIT V1` board on Arduino IDE `1.8.10` I am getting conflicts with the "default" Arduino servo library.

Due to this problem, none of the **examples** compiles.

This is easily fixed by changing the name of the header file to something more unique. Ideally, Arduino IDE would have a way of differentiating them, but yeah, it doesn't.

I suggest by renaming `Servo.h` :arrow_right: `ESP32ArduinoServo.h`. Based on the name you have chosen for your repository, this sounds like a reasonable name to me.

ruiseixasm, 2020-01-15:
Still gives this error! Why don't the actor simple gives a non conflicting name? Simple puzzling!...

Arduino: 1.8.10 (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

WARNING: library Servo claims to run on avr, megaavr, sam, samd, nrf52, stm32f4, mbed architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
In file included from C:\Users\Utilizador\Documents\Arduino\libraries\ESP32-Arduino-Servo-Library-master\src\ESP32ArduinoServo.cpp:33:0:

C:\Users\Utilizador\Documents\Arduino\libraries\Servo\src/Servo.h:77:2: error: #error "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."

 #error "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."

  ^

C:\Users\Utilizador\Documents\Arduino\libraries\ESP32-Arduino-Servo-Library-master\src\ESP32ArduinoServo.cpp:35:12: error: 'int Servo::channel_next_free' is not a static data member of 'class Servo'

 int Servo::channel_next_free = 0;

            ^

C:\Users\Utilizador\Documents\Arduino\libraries\ESP32-Arduino-Servo-Library-master\src\ESP32ArduinoServo.cpp: In constructor 'Servo::Servo()':

C:\Users\Utilizador\Documents\Arduino\libraries\ESP32-Arduino-Servo-Library-master\src\ESP32ArduinoServo.cpp:38:18: error: '_resetFields' was not declared in this scope

     _resetFields();

                  ^

C:\Users\Utilizador\Documents\Arduino\libraries\ESP32-Arduino-Servo-Library-master\src\ESP32ArduinoServo.cpp: At global scope:

C:\Users\Utilizador\Documents\Arduino\libraries\ESP32-Arduino-Servo-Library-master\src\ESP32ArduinoServo.cpp:41:15: error: definition of implicitly-declared 'Servo::~Servo()'

 Servo::~Servo() {

               ^

C:\Users\Utilizador\Documents\Arduino\libraries\ESP32-Arduino-Servo-Library-master\src\ESP32ArduinoServo.cpp:45:6: error: prototype for 'bool Servo::attach(int, int, int, int, int, int)' does not match any in class 'Servo'

 bool Servo::attach(int pin, int channel, 

      ^

In file included from C:\Users\Utilizador\Documents\Arduino\libraries\ESP32-Arduino-Servo-Library-master\src\ESP32ArduinoServo.cpp:33:0:

C:\Users\Utilizador\Documents\Arduino\libraries\Servo\src/Servo.h:109:11: error: candidates are: uint8_t Servo::attach(int, int, int)

   uint8_t attach(int pin, int min, int max); // as above but also sets min and max values for writes. 

           ^

In file included from C:\Users\Utilizador\Documents\Arduino\libraries\ESP32-Arduino-Servo-Library-master\src\ESP32ArduinoServo.cpp:33:0:

C:\Users\Utilizador\Documents\Arduino\libraries\Servo\src/Servo.h:108:11: error:                 uint8_t Servo::attach(int)

   uint8_t attach(int pin);           // attach the given pin to the next free channel, sets pinMode, returns channel number or 0 if failure

           ^

C:\Users\Utilizador\Documents\Arduino\libraries\ESP32-Arduino-Servo-Library-master\src\ESP32ArduinoServo.cpp:71:6: error: prototype for 'bool Servo::detach()' does not match any in class 'Servo'

 bool Servo::detach() {

      ^

In file included from C:\Users\Utilizador\Documents\Arduino\libraries\ESP32-Arduino-Servo-Library-master\src\ESP32ArduinoServo.cpp:33:0:

C:\Users\Utilizador\Documents\Arduino\libraries\Servo\src/Servo.h:110:8: error: candidate is: void Servo::detach()

   void detach();

        ^
[... 113 lines cut ...]
        ^

C:\Users\Utilizador\Documents\Arduino\libraries\ESP32-Arduino-Servo-Library-master\src\ESP32ArduinoServo.cpp:111:26: error: no 'int Servo::attachedPin() const' member function declared in class 'Servo'

 int Servo::attachedPin() const { return _pin; }

                          ^

C:\Users\Utilizador\Documents\Arduino\libraries\ESP32-Arduino-Servo-Library-master\src\ESP32ArduinoServo.cpp:113:30: error: no 'void Servo::_resetFields()' member function declared in class 'Servo'

 void Servo::_resetFields(void) {

                              ^

Multiple libraries were found for "ESP32ArduinoServo.h"
 Used: C:\Users\Utilizador\Documents\Arduino\libraries\ESP32-Arduino-Servo-Library-master
Multiple libraries were found for "Servo.h"
 Used: C:\Users\Utilizador\Documents\Arduino\libraries\Servo
 Not used: C:\Program
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

JarekParal, 2020-01-22:
Hi,
I tested the library and found that the problem happens just with the latest version of Arduino IDE (1.8.10).
With the versions, 1.8.9 and 1.8.8 everything works.

I created an issue on the Arduino GitHub repository to get some more information about the change between the last version: https://github.com/arduino/Arduino/issues/9655

Actually I don't want to change the name of the files `Servo.h` and `Servo.cpp` because this is breaking change which affects all users of this library.
And because this library is used mainly with PlatformIO, [where is no issue with build](https://travis-ci.org/RoboticsBrno/ESP32-Arduino-Servo-Library), I would like to rather fix this issue in Arduino IDE.

But if you need urgently some library for servo on ESP32 I found two similar libraries:
- https://github.com/khoih-prog/ESP32_ISR_Servo
- https://github.com/jkb-git/ESP32Servo

platisd, 2020-01-22:
Interesting that there is no issue with the previous IDE versions. I guess it has to do with the priority the libraries are chosen.

I have looked into the two libraries you mentioned and unfortunately they have other problems. The first one uses a different API than the Arduino one (which should not be that much of a problem really, but if I can avoid code changes, why not?) and the second, is published via the GPLv2.1 license which I want to avoid to be able to keep publishing [my library](https://github.com/platisd/smartcar_shield) with a permissive license.

Finally, it is perfectly understandable you do not want to break compatibility for existing users. So unless Arduino IDE manages to fix this on their end soon enough, if you do not mind, I will fork the project, make the changes and publish the Arduino IDE-compatible version on the library manager there, always attributing the original work of course. :+1:

JarekParal, 2020-01-23:
Yes, I haven't got any problem with the fork and creating your own version but could you wait some time with the publication in Arduino IDE Library Manager. I hope that I will get soon some information from Arduino IDE team and find the way how to deal with this issue.
I would be like if we won't fritter this library to many forks and each fork will be available on different platform (Arduino IDE vs PlatformIO). 
Could we give Arduino Team two weeks to get some info about this problem after that decide what will be the next step?

PS: I also tested changing the names of the files from `Servo{.h/.cpp}` to `ServoESP32{.h/.cpp}` and it seems that this fixed the error in Arduino IDE 1.8.10. I used this name on [PlatformIO](http://platformio.org/lib/show/1739/ServoESP32).

platisd, 2020-01-23:
Sure, I can wait until then, since I'm waiting for the Spring Festival in China to finish so I can order some essential components. This is not blocking me at this point.

I'm OK with the `ESP32Servo` naming. I didn't pick it since there's another library having a header with the same name but if it's fine with you it's fine with me.

JarekParal, 2020-01-23:
The bug with the import is already fixed in the Arduino IDE. The new version going to be released on Monday, Jan, 27.

https://github.com/arduino/arduino-cli/pull/565

Then we can test this library again.

platisd, 2020-01-23:
Sounds awesome! Thanks for looking into this! :)

I'm curious though, what happens if someone has the library installed and want to compile for another architecture where the default `Servo.h` should be used? Wouldn't this library be erroneously preferred over the correct one in that case?

JarekParal, 2020-01-23:
Yes, according the code snippet which you sent there is some handling of the libraries and the original library from Arduino is prefered.
I also tested this case in Arduino IDE 1.8.9 and when I build one of my servo example for Arduino Nano, the compiler select original library from Arduino:
```sh
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\noname\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\noname\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\noname\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10809 -build-path C:\Users\noname\AppData\Local\Temp\arduino_build_517251 -warnings=all -build-cache C:\Users\noname\AppData\Local\Temp\arduino_cache_369272 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\noname\Documents\Arduino\libraries\ESP32-Arduino-Servo-Library\examples\02-ServoPotentiometer\02-ServoPotentiometer.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\noname\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\noname\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\noname\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10809 -build-path C:\Users\noname\AppData\Local\Temp\arduino_build_517251 -warnings=all -build-cache C:\Users\noname\AppData\Local\Temp\arduino_cache_369272 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\noname\Documents\Arduino\libraries\ESP32-Arduino-Servo-Library\examples\02-ServoPotentiometer\02-ServoPotentiometer.ino
Using board 'nano' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
WARNING: Spurious .pio folder in 'ServoESP32' library
Build options changed, rebuilding all
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "C:\\Users\\noname\\AppData\\Local\\Temp\\arduino_build_517251\\sketch\\02-ServoPotentiometer.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IC:\\Program Files (x86)\\Arduino\\libraries\\Servo\\src" "C:\\Users\\noname\\AppData\\Local\\Temp\\arduino_build_517251\\sketch\\02-ServoPotentiometer.ino.cpp" -o nul

...

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-gcc-ar" rcs "C:\\Users\\noname\\AppData\\Local\\Temp\\arduino_build_517251\\core\\core.a" "C:\\Users\\noname\\AppData\\Local\\Temp\\arduino_build_517251\\core\\wiring_pulse.c.o"
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-gcc-ar" rcs "C:\\Users\\noname\\AppData\\Local\\Temp\\arduino_build_517251\\core\\core.a" "C:\\Users\\noname\\AppData\\Local\\Temp\\arduino_build_517251\\core\\wiring_shift.c.o"
Archiving built core (caching) in: C:\Users\noname\AppData\Local\Temp\arduino_cache_369272\core\core_arduino_avr_nano_cpu_atmega328_0c812875ac70eb4a9b385d8fb077f54c.a
Linking everything together...
[... cut ...]
 Not used: C:\Users\noname\Documents\Arduino\libraries\ESP32-Arduino-Servo-Library
Using library Servo at version 1.1.3 in folder: C:\Program Files (x86)\Arduino\libraries\Servo 
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-size" -A "C:\\Users\\noname\\AppData\\Local\\Temp\\arduino_build_517251/02-ServoPotentiometer.ino.elf"
Sketch uses 3278 bytes (10%) of program storage space. Maximum is 30720 bytes.
Global variables use 229 bytes (11%) of dynamic memory, leaving 1819 bytes for local variables. Maximum is 2048 bytes.
```

platisd, 2020-01-23:
Nice! It should work seamlessly then! :+1:

platisd, 2020-01-28:
Verified the issue is no more with Arduino IDE 1.8.11 on Linux :+1:

closed 2020-01-28 by platisd