RoboticsBrno/ServoESP32 #14: Can't use library with MG 996R servos

sensboston, 2020-12-13:
This library works perfectly with regular servos, like SG90, but when I tried high torque MG996R, behavior is stochastic. I know, MG996R is **continuous** servo but motor don't want "behave" even as [described here](https://arduino.stackexchange.com/questions/1321/servo-wont-stop-rotating) (in answer).

Any suggestions guys? Or please share a link to the **standard** high torque servos, not expensive as a hell 😉

JarekParal, 2021-11-21:
Could you share a detailed description of your stochastic behaviour? Ideally with example code, which you used for testing?

@yaqwsx Do you have some suggestions: "standard high torque servos, not expensive as a hell"

yaqwsx, 2021-11-22:
If you want to go cheap and pretty good quality, look at Lewansoul servos.

1248280302, 2022-04-17:
Is this problem solved please?

JarekParal, 2022-04-19:
We didn't get more info about the issue from @sensboston -> didn't do anything with this issue. @1248280302  Do you have a similar problem?

closed 2023-04-04 by JarekParal