RoboticsBrno/ServoESP32 #16: 'ledcSetup' was not declared in this scope

mdmbct, 2021-11-07:
I use this lib in PlatformIO for CLion, but the console print these msgs shown in the img.

JarekParal, 2021-11-07:
Hi, the `ledcXY` methods should be part of the Arduino ESP32 framework included with `#include "Arduino.h"`. Could you check that the framework is properly configured? 
Could you also copy/paste the whole build log?
We are using Platform.io mainly in VSCode. I don't know if you need to do something special in CLion to make it work. Please also share your `platformio.ini` file.

mdmbct, 2021-11-07:
> Hi, the `ledcXY` methods should be part of the Arduino ESP32 framework included with `#include "Arduino.h"`. Could you check that the framework is properly configured? Could you also copy/paste the [...]

As a beginner, I didn't distinguish the esp32 and esp8266🥲. The lib is for esp32 but my dk is esp8266🤦‍♂️🤦‍♂️.

closed 2021-11-07 by mdmbct