RoboticsBrno/ServoESP32 #18: Compatibility with ESP32-S2

f-peri, 2022-09-03:
Hello, thank you for developing this library. I am using the ESP32-S2 and from expressif's documentation, the PWM's are limited to 14-bits. As written, I have not been able to get the code to work on an S2, however, once I changed the ledcSetup call to 14, it works well.

From:
`ledcSetup(_channel, 50, 16); // channel X, 50 Hz, 16-bit depth`

To:
`ledcSetup(_channel, 50, 14); // channel X, 50 Hz, 14-bit depth`

You may want to consider with passing this as a variable or creating a function that allows this to be set. Thanks again.

JarekParal, 2023-04-04:
@Tasssadar Do you think you can help me with this issue? You have more experience in specific ESP32 modules modification. Is there any define or macro which we can use to check which ESP32 module is used?

Tasssadar, 2023-04-04:
Yeah, you can use defines like `CONFIG_IDF_TARGET_ESP32S3`: https://github.com/RoboticsBrno/SmartLeds/blob/7e92147351c969b16513b3bf39db2f65b6c8811a/src/SmartLeds.h#L186-L192

JarekParal, 2023-04-13:
There is information that ESP32-S2 has an 18-bits LED PWM controller in their datasheet (page 26): https://www.espressif.com/sites/default/files/documentation/esp32-s2_datasheet_en.pdf

Then I don't understand why it didn't work with 16-bits. 

@f-peri Do you think you can test the code with 18-bits?
```cpp
ledcSetup(_channel, 50, 18); // channel X, 50 Hz, 18-bits depth
```

f-peri, 2023-04-14:
@JarekParal It's been awhile since I've worked on that project, but I'll pull it up and review. I see your point in the ref document: "The accuracy of duty can be up to 18 bits within a 1 ms period." though that wording is a bit vague. I'll have to look back at my notes to find where I saw 14-bit documented. Thanks again for investigating.

mikaeltulldahl, 2023-05-01:
Seems like @h2zero has solved it here: https://github.com/h2zero/ServoESP32/tree/s3

mikaeltulldahl, 2023-08-29:
Should be solved with latest release 1.1.0

closed 2023-08-30 by JarekParal