RoboticsBrno/ServoESP32 #21: servo YUKI model family SUMO 1143HB

fotosettore, 2023-04-03:
hi 
this servo (SUMO 1143HB)  has a mechanical 360 degree rotation.
using this library is not possible to have more than 180 defgree
for my use I need a 260 degree rotation
I try to modify the .h and .cpp but i'm not smart in c++, so at moment no solution.
someone may help me to do this ?
I saw a lot of video but all them make a hardware modification.
I'm sure that with this kind of servo is possible to obtain 260 degree via software too.
Please look the video that solve the problem without opening of servo
https://www.aliexpress.com/item/1005002973104855.html?spm=a2g0o.order_list.order_list_main.19.21ef1802XnKwfL

your help is appreciated

JarekParal, 2023-04-04:
Hello @fotosettore,

I checked the spec on Aliexpress and the recommended PWM signal range which should be 0.5 - 2.5 which is exactly the range in our library: https://github.com/RoboticsBrno/ServoESP32/blob/master/src/Servo.h#L107

But you can try to extend the range from 0.1 to 3.0 ms and see what would be the impact:
```cpp
Servo servo1;

void setup() {
    Serial.begin(115200);
    int chanell = 0;
    int minAngle = 0;
    int maxAngle = 180; 
    // the `maxAngle` doesn't have direct effect on the PWM signal, 
    // just divides the range 2900 (3000 - 100) to 180 degree steps in `write(int degrees)` function
    int minPulseWidth = 100;
    int maxPulseWidth = 3000;

    servo1.attach(servoPin, channel, minAngle, maxAngle, minPulseWidth, maxPulseWidth);
}
...
```

fotosettore, 2023-04-10:
hi JarekParal
i tried your code and i obtained only little degree more.
So I opened the servo and added two resistors as described in a youtube video.
the two solutions (your code and resistors) improved all but I didn't get the result :-(
however i'd like to goal without hardware modifications.
So I've come to the conclusion that for 260° a native servo is needed.
I bought some of them and as soon as they will arrive I will try and I will post here the test made using your library

JarekParal, 2023-04-13:
Thanks for update. Let me know when you will have new info.