RoboticsBrno/ServoESP32 #26: Could not build library with provided examples

hstag, 2023-10-01:
Hi, I am trying to use this library on ESP32, but when I try to build the example I get more errors:

for example:
`In file included from src/main.cpp:2:
.pio/libdeps/esp32dev/ServoESP32/src/Servo.h:68:81: error: call to non-'constexpr' function 'const _Tp& std::min(const _Tp&, const _Tp&) [with _Tp = int]'
     static const int TIMER_RESOLUTION = std::min(16, SOC_LEDC_TIMER_BIT_WIDE_NUM);`

Log, platformio.ini and code included.

[attachments.zip](https://github.com/RoboticsBrno/ServoESP32/files/12778251/attachments.zip)

Roemke, 2023-10-10:
Hi, I have the same problem, 
first I thought: ok my compiler uses -std=gnu++11 and i changed it in platform.local.txt  
using linux: grep "gnu++" platform.txt > platform.local.txt
and then change it with an editor (4 times) to gnu++17.
It does still not work, but it seems to use again gnu++11 ? I don't understand. 
Furthermore some just compiled files are used, I don't know if that's a problem. 
I just removed /tmp/arduino* where old files are stored. 
After that I can compile a program which uses Servo.h but it does not work anymore ...
So I have no success, but downgrading to 1.0.3 or 1.0.2 which works.

jamie-54, 2023-10-27:
I've had the same issue so downgrading to 1.0.3 which works.

ArduinoGetStarted, 2023-11-16:
Dear @JarekParal 

Thank you for your great library.
Do you have a plan to release new version that fix this issue ?

JarekParal, 2023-11-16:
Hi @ArduinoGetStarted,

yes, I would like to fix the bug in the coming month. I was very overloaded with a work until now, but it should be better now. 

Still, fix PR from anybody would be welcome.

JarekParal, 2023-11-16:
In case you would need to use the latest version (1.1.1) you can use it with PlatformIO IDE.

In `platformio.ini` you need to set the compiler to `gnu++17`:
```ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
build_unflags = -std=gnu++11
build_flags = -std=gnu++17

lib_deps =
  roboticsbrno/ServoESP32 @ 1.1.1
```

With this `platformio.ini` was able to compile the first example:
```cpp
#include <Arduino.h>
#include <Servo.h>

static const int servoPin = 4;

Servo servo1;

void setup()
{
  Serial.begin(115200);
  servo1.attach(servoPin);
}

void loop()
{
  for (int posDegrees = 0; posDegrees <= 180; posDegrees++)
  {
    servo1.write(posDegrees);
    Serial.println(posDegrees);
    delay(20);
  }

  for (int posDegrees = 180; posDegrees >= 0; posDegrees--)
  {
    servo1.write(posDegrees);
    Serial.println(posDegrees);
    delay(20);
  }
}
```

JarekParal, 2023-11-16:
> Hi, I have the same problem, first I thought: ok my compiler uses -std=gnu++11 and i changed it in platform.local.txt using linux: grep "gnu++" platform.txt > platform.local.txt and then change it [...]

@Roemke 
I was able to compile the `ServoESP32 v1.1.1` with `Arduino IDE 1.8.19` when I replaced the `gnu++11` with `gnu++17` in this file:
C:\Users\paral\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\platform.txt

(same ServoESP32 example as in the comment above)

That is not a proper fix/solution, but at least it is a workaround. 

Please let me know if that would work for you.

per1234, 2023-12-20:
@JarekParal since it appears you don't use the official Arduino development software and thus won't be able to manually detect the introduction of changes that break the library for users of that software, I recommend you to set up an automated compilation of the library examples. This can be done by adding a [**GitHub Actions** workflow](https://docs.github.com/actions/using-workflows/about-workflows) to the repository.

Arduino provides a GitHub Actions action specifically for this purpose, which is used in all the official Arduino repositories. Using that action in the workflow should make it quite easy to set up the system:

https://github.com/arduino/compile-sketches

JarekParal, 2023-12-20:
Hi @per1234, thanks for your suggestion.

Long time ago, we set up TravisCI which built the examples. But this is not working anymore and doesn't make sense to fix it.

I have already tried to set up GitHub Actions for this repository a few months ago. But I was unable to get a workable version in the timeframe which I allocated to this task.
You can take a look on this branch and give me feedback how to fix the build: https://github.com/RoboticsBrno/ServoESP32/tree/feature/add-github-workflow

Or any PR for this would be really welcome.

rsuzano, 2024-01-21:
> Hi @per1234, thanks for your suggestion.
> 
> A long time ago, we set up TravisCI which built the examples. But this is not working anymore and doesn't make sense to fix it.
> 
> I have already tried to set up GitHub Actions for this repository a few months ago. But I was unable to get a workable version in the timeframe which I allocated to this task. You can take a look at this branch and give me feedback on how to fix the build: https://github.com/RoboticsBrno/ServoESP32/tree/feature/add-github-workflow
> 
> Or any PR for this would be really welcome.
> 
> For traceability, I created new issue #29.

Hi @JarekParal I have this repository with working github actions build... 
https://github.com/rsuzano/OnStepX/blob/69ec3f015836a41a7ce8d06c3044f036cd2caa14/.github/workflows/build.yml

If you allow me I can try to port this action to a PR in your repository, just let me know what you think about it.

JarekParal, 2024-01-21:
> Hi @JarekParal I have this repository with working github actions build... https://github.com/rsuzano/OnStepX/blob/69ec3f015836a41a7ce8d06c3044f036cd2caa14/.github/workflows/build.yml
>
> If you allow [...]

Hi @rsuzano, yes your help would be really appreciated. If you would be able to create working GitHub Action to compile the example files with Arduino IDE and/or with PlatformIO, then we should minimize the issues with builds and PR merges. 

When you will have something to share don't hesitate to open a PR. 

Thanks in advance.

zyphlar, 2024-04-27:
Thanks for this library. Replacing all occurrences of `gnu++11` with `gnu++17` results in the same error on Arduino IDE 2.3.2 for an ESP32 QWIIC board from SparkFun.

```
name=ESP32 Arduino
version=

tools.xtensa-esp32-elf-gcc.path={runtime.tools.xtensa-esp32-elf-gcc.path}
tools.xtensa-esp32s2-elf-gcc.path={runtime.tools.xtensa-esp32s2-elf-gcc.path}
tools.xtensa-esp32s3-elf-gcc.path={runtime.tools.xtensa-esp32s3-elf-gcc.path}
tools.xtensa-esp-elf-gdb.path={runtime.tools.xtensa-esp-elf-gdb.path}
tools.riscv32-esp-elf-gcc.path={runtime.tools.riscv32-esp-elf-gcc.path}
tools.riscv32-esp-elf-gdb.path={runtime.tools.riscv32-esp-elf-gdb.path}

debug.server.openocd.path={runtime.tools.openocd-esp32.path}/bin/openocd
debug.server.openocd.scripts_dir={runtime.tools.openocd-esp32.path}/share/openocd/scripts/
debug.server.openocd.scripts_dir.windows={runtime.tools.openocd-esp32.path}\share\openocd\scripts\

tools.esptool_py.path={runtime.tools.esptool_py.path}
tools.esptool_py.cmd=esptool
tools.esptool_py.cmd.linux=esptool.py
tools.esptool_py.cmd.windows=esptool.exe

tools.esptool_py.network_cmd=python3 "{runtime.platform.path}/tools/espota.py" -r
tools.esptool_py.network_cmd.windows="{runtime.platform.path}/tools/espota.exe" -r

tools.esp_ota.cmd=python3 "{runtime.platform.path}/tools/espota.py" -r
tools.esp_ota.cmd.windows="{runtime.platform.path}/tools/espota.exe" -r

tools.gen_esp32part.cmd=python3 "{runtime.platform.path}/tools/gen_esp32part.py"
tools.gen_esp32part.cmd.windows="{runtime.platform.path}/tools/gen_esp32part.exe"

tools.gen_insights_pkg.cmd=python3 "{runtime.platform.path}"/tools/gen_insights_package.py
tools.gen_insights_pkg.cmd.windows="{runtime.platform.path}/tools/gen_insights_package.exe"

compiler.path={tools.{build.tarch}-{build.target}-elf-gcc.path}/bin/
compiler.sdk.path={runtime.platform.path}/tools/sdk/{build.mcu}
compiler.prefix={build.tarch}-{build.target}-elf-

#
# ESP32 Support Start
#
compiler.cpreprocessor.flags.esp32=-DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE -DIDF_VER="v4.4.5" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS   "-I{compiler.sdk.path}/include/newlib/platform_include" "-I{compiler.sdk.path}/include/freertos/include" "-I{compiler.sdk.path}/include/freertos/include/esp_additions/freertos" "-I{compiler.sdk.path}/include/freertos/port/xtensa/include" "-I{compiler.sdk.path}/include/freertos/include/esp_additions" "-I{compiler.sdk.path}/include/esp_hw_support/include" "-I{compiler.sdk.path}/include/esp_hw_support/include/soc" "-I{compiler.sdk.path}/include/esp_hw_support/include/soc/esp32" "-I{compiler.sdk.path}/include/esp_hw_support/port/esp32" "-I{compiler.sdk.path}/include/esp_hw_support/port/esp32/private_include" "-I{compiler.sdk.path}/include/heap/include" "-I{compiler.sdk.path}/include/log/include" "-I{compiler.sdk.path}/include/lwip/include/apps" "-I{compiler.sdk.path}/include/lwip/include/apps/sntp" "-I{compiler.sdk.path}/include/lwip/lwip/src/include" "-I{compiler.sdk.path}/include/lwip/port/esp32/include" "-I{compiler.sdk.path}/include/lwip/port/esp32/include/arch" "-I{compiler.sdk.path}/include/soc/include" "-I{compiler.sdk.path}/include/soc/esp32" "-I{compiler.sdk.path}/include/soc/esp32/include" "-I{compiler.sdk.path}/include/hal/esp32/include" "-I{compiler.sdk.path}/include/hal/include" "-I{compiler.sdk.path}/include/hal/platform_port/include" "-I{compiler.sdk.path}/include/esp_rom/include" "-I{compiler.sdk.path}/include/esp_rom/include/esp32" "-I{compiler.sdk.path}/include/esp_rom/esp32" "-I{compiler.sdk.path}/include/esp_common/include" "-I{compiler.sdk.path}/include/esp_system/include" "-I{compiler.sdk.path}/include/esp_system/port/soc" "-I{compiler.sdk.path}/include/esp_system/port/public_compat" "-I{compiler.sdk.path}/include/esp32/include" "-I{compiler.sdk.path}/include/xtensa/include" "-I{compiler.sdk.path}/include/xtensa/esp32/include" "-I{compiler.sdk.path}/include/driver/include" "-I{compiler.sdk.path}/include/driver/esp32/include" "-I{compiler.sdk.path}/include/esp_pm/include" "-I{compiler.sdk.path}/include/esp_ringbuf/include" "-I{compiler.sdk.path}/include/efuse/include" "-I{compiler.sdk.path}/include/efuse/esp32/include" "-I{compiler.sdk.path}/include/vfs/include" "-I{compiler.sdk.path}/include/esp_wifi/include" "-I{compiler.sdk.path}/include/esp_event/include" "-I{compiler.sdk.path}/include/esp_netif/include" "-I{compiler.sdk.path}/include/esp_eth/include" "-I{compiler.sdk.path}/include/tcpip_adapter/include" "-I{compiler.sdk.path}/include/esp_phy/include" "-I{compiler.sdk.path}/include/esp_phy/esp32/include" "-I{compiler.sdk.path}/include/esp_ipc/include" "-I{compiler.sdk.path}/include/app_trace/include" "-I{compiler.sdk.path}/include/esp_timer/include" "-I{compiler.sdk.path}/include/mbedtls/port/include" "-I{compiler.sdk.path}/include/mbedtls/mbedtls/include" "-I{compiler.sdk.path}/include/mbedtls/esp_crt_bundle/include" "-I{compiler.sdk.path}/include/app_update/include" "-I{compiler.sdk.path}/include/spi_flash/include" "-I{compiler.sdk.path}/include/bootloader_support/include" "-I{compiler.sdk.path}/include/nvs_flash/include" "-I{compiler.sdk.path}/include/pthread/include" "-I{compiler.sdk.path}/include/esp_gdbstub/include" "-I{compiler.sdk.path}/include/esp_gdbstub/xtensa" "-I{compiler.sdk.path}/include/esp_gdbstub/esp32" "-I{compiler.sdk.path}/include/espcoredump/include" "-I{compiler.sdk.path}/include/espcoredump/include/port/xtensa" "-I{compiler.sdk.path}/include/ [... cut ...]
tools.esp_ota.upload.pattern={cmd} -i {upload.port.address} -p {upload.port.properties.port} --auth={upload.field.password} -f "{build.path}/{build.project_name}.bin"

## Upload Sketch Through DFU OTA
## -------------------------------------------
tools.dfu-util.path={runtime.tools.dfu-util-0.11.0-arduino5.path}
tools.dfu-util.cmd=dfu-util
tools.dfu-util.upload.params.verbose=-d
tools.dfu-util.upload.params.quiet=
tools.dfu-util.upload.pattern="{path}/{cmd}" --device {vid.0}:{pid.0} -D "{build.path}/{build.project_name}.bin" -Q
```

zyphlar, 2024-04-28:
This "fixed" library does compile however https://github.com/alunit3/ServoESP32/