RoboticsBrno/ServoESP32 #30: Compilling error

rsuzano, 2024-01-20:
Hi, fist of all, thx for your lib!

I can't build my sketch because of these errors, I'm  using PlatformIO, but I've tested ArduinoIDE and received diferent erro on the same file and line of code:

**PlatformIO**
Executing task in folder Nykko_Brat_rsuzano: platformio run --target upload 

Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (5.2.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5) 
 - tool-esptoolpy @ 1.40201.0 (4.2.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- DabbleESP32 @ 1.5.1
|-- PWMMotorControl @ 2.1.0
|-- ESP32 AnalogWrite @ 0.2.0
|-- ServoESP32 @ 1.1.1
|-- Preferences @ 2.0.0
Building in release mode
Compiling .pio/build/esp32dev/src/main.cpp.o
In file included from src/main.h:5,
                 from src/main.cpp:1:
.pio/libdeps/esp32dev/ServoESP32/src/Servo.h:68:81: error: call to non-'constexpr' function 'const _Tp& std::min(const _Tp&, const _Tp&) [with _Tp = int]'
     static const int TIMER_RESOLUTION = std::min(16, SOC_LEDC_TIMER_BIT_WIDE_NUM);
                                                                                 ^
.pio/libdeps/esp32dev/ServoESP32/src/Servo.h: In member function 'T ServoTemplate<T>::mapTemplate(T, T, T, T, T) const':
.pio/libdeps/esp32dev/ServoESP32/src/Servo.h:256:12: warning: 'if constexpr' only available with -std=c++17 or -std=gnu++17
         if constexpr (std::is_floating_point_v<T>) {
            ^~~~~~~~~
.pio/libdeps/esp32dev/ServoESP32/src/Servo.h:256:28: error: 'is_floating_point_v' is not a member of 'std'
         if constexpr (std::is_floating_point_v<T>) {
                            ^~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32dev/ServoESP32/src/Servo.h:256:28: note: suggested alternative: 'is_floating_point'
         if constexpr (std::is_floating_point_v<T>) {
                            ^~~~~~~~~~~~~~~~~~~
                            is_floating_point
.pio/libdeps/esp32dev/ServoESP32/src/Servo.h:256:49: error: expected primary-expression before '>' token
         if constexpr (std::is_floating_point_v<T>) {
                                                 ^
.pio/libdeps/esp32dev/ServoESP32/src/Servo.h:256:50: error: expected primary-expression before ')' token
         if constexpr (std::is_floating_point_v<T>) {
                                                  ^
.pio/libdeps/esp32dev/ServoESP32/src/Servo.h: In member function 'T ServoTemplate<T>::mapTemplate(T, T, T, T, T) const [with T = int]':
.pio/libdeps/esp32dev/ServoESP32/src/Servo.h:262:5: warning: control reaches end of non-void function [-Wreturn-type]

**ArduinoIDE**
In file included from /home/rsuzano/Arduino/libraries/ServoESP32/examples/01-SimpleServo/01-SimpleServo.ino:1:0:
/home/rsuzano/Arduino/libraries/ServoESP32/src/Servo.h:68:54: error: 'SOC_LEDC_TIMER_BIT_WIDE_NUM' was not declared in this scope
     static const int TIMER_RESOLUTION = std::min(16, SOC_LEDC_TIMER_BIT_WIDE_NUM);
                                                      ^
/home/rsuzano/Arduino/libraries/ServoESP32/src/Servo.h: In member function 'bool ServoTemplate<T>::attach(int, int, T, T, int, int, int)':
/home/rsuzano/Arduino/libraries/ServoESP32/src/Servo.h:52:24: error: 'SOC_LEDC_CHANNEL_NUM' was not declared in this scope
 #define LEDC_CHANNELS (SOC_LEDC_CHANNEL_NUM)
                        ^
/home/rsuzano/Arduino/libraries/ServoESP32/src/Servo.h:132:38: note: in expansion of macro 'LEDC_CHANNELS'
             if (channel_next_free == LEDC_CHANNELS) {
                                      ^
/home/rsuzano/Arduino/libraries/ServoESP32/src/Servo.h: In member function 'T ServoTemplate<T>::mapTemplate(T, T, T, T, T) const':
/home/rsuzano/Arduino/libraries/ServoESP32/src/Servo.h:256:12: error: expected '(' before 'constexpr'
         if constexpr (std::is_floating_point_v<T>) {
            ^
/home/rsuzano/Arduino/libraries/ServoESP32/src/Servo.h:258:11: error: 'else' without a previous 'if'
         } else {

JarekParal, 2024-01-21:
Hi @rsuzano, I think you have same problem as we discussed in #26. Could you please check this comment and try to use the proposed solution (until I will be able to provide final solution in next library release): https://github.com/RoboticsBrno/ServoESP32/issues/26#issuecomment-1813975109

Let me please know, if that would solve your issue.

rsuzano, 2024-01-21:
Hi, sorry for not find the issue #26 before open this. I'll try that solution. 
Thanks