RoboticsBrno/ServoESP32 #31: Frequency parameter is not working as expected

reivaxy, 2024-03-20:
Hi and thanks for this library.

It seems like the frequency parameter that can be passed to the attach function is not actually changing the frequency but the pulse width.
I have initialized two Servos like this: 

	  myservo1.attach(servoPin1, Servo::CHANNEL_NOT_ATTACHED, Servo::DEFAULT_MIN_ANGLE,
                Servo::DEFAULT_MAX_ANGLE, Servo::DEFAULT_MIN_PULSE_WIDTH_US,
                Servo::DEFAULT_MAX_PULSE_WIDTH_US, 200);

	  myservo2.attach(servoPin2);

Here are signals measured on two pins, the first (yellow) one was set at 200Hz, and the second (blue) one was left at default value which is 50Hz
And both signals are actually at 50Hz
[image]

Here is what is expected: the yellow signal has a frequency 4 times higher:
[image]

reivaxy, 2024-03-21:
Also tested with frequencies 200 on pin1 and 400 Hz on pin2 and both pins show a 400 frequency.
If I reverse and set 400 on 1 and 200 on 2, both have a 200Hz signal: the last pin attached set the frequency for all.

thijstriemstra, 2024-04-16:
what board are you using?

reivaxy, 2024-04-16:
XIAO ESP32-S3